vrolife/modern_laptop #89: Arch Linux boot crash without charger plugged

MalpenZibo, 2023-09-01:
I notice that if the battery is under 70% and my charger is not plugged in, arch Linux crash during boot just before asking for my password to decrypt the partition.

With charger plugged in or with battery over 70% the boot process works without any issue.

Model: TM2119-Xiaomi_Book_Pro_14_2022

Someone has the same problem or a solution?

eukhlg, 2023-09-07:
Hello 
I have the same problem and I'm aware that some other users of Xiaomi_Book_Pro_14_2022 have one. Unfortunately no solution so far. I believe this is some firmware bug and only can be fixed in new BIOS release. I also noticed that when running with acpi=off the issue is not occur.

MalpenZibo, 2023-12-24:
About this topic there's no valid solution?
At this page https://www.mi.com/service/notebook/drivers/A54S I can find my current bios version Xiaomi book 14 pro oled 2022 Ryzen edition 

In this page https://www.mi.com/service/notebook/drivers/A28S it seems present a recent bios this end with 0E0E the other with 0A0A but it's for the Xiaomi book 14 pro 2022 Ryzen edition.

Somebody knows if the other bios can be used and if resolve the issue?