vrolife/modern_laptop #91: Sleep problem on Redmibook Pro 14 2022 Intel (12650h)

dgefremov, 2023-11-19:
Description:
On Redmibook 2022 intel edition (14 or 15 inch models, i5-12450h and i7-12650h models) there is a bug: after suspending and resuming there is no screen and keyboard backlight on linux. After resuming using flashlight I can see ssdm screen, fans are working, volume adjust is working too. There are no errors in `dmesg`. 
On windows everything is ok. 
What I have already tried:
0) Different kernels (from 6.1 up to 6.5)
1) Different kernel parameters(`osi=! osi=Linux`, `osi=! osi=Windows 2015`, `acpi_backlight=native\video\vendor`);
2) Trying to adjust brightness through `sysfs` after resume;
3) Tryint to adjust brighness through `acpi_exec` (calling `_BCM` method);
4) Decompinling dsdt\ssdt files, but not errors found. One ssdt table doesn't dump because of wrong header, but I don't think it is the reason of broken sleep;
5) Trying S3 sleep mode (it is avaliable by default on intel models), but after resume system hangs: black screen, althouth keyboard backlight is working.
6) Googling everywhere :)
Sometimes laptop resumes correctly (one time of 10-20).
Do you have any ideas where the bug could be? I'm really in stuck.

ThatOneCalculator, 2024-01-23:
I have the same issue on a Redmibook Pro 15S AMD (RMACZ5B0P0909). Did you find a fix?

dgefremov, 2024-01-26:
Nope. I think problem is somewhere deep in SSDT, maybe problems with backlight power supply initialization (due to race condition). So I set up hibernation on lid close event as a workaround, but it is not so comfortable as it could be...

ThatOneCalculator, 2024-01-26:
Could you tell me how to set up hibernation on lid close?

malankaMcQueen, 2026-11-15:
Thank you!!!! You solved my problem!!!