vrolife/modern_laptop #93: Will it work on Redibook pro 15 2023 Ryzen 7 7840HS?

sasha0000kr, 2023-11-29:


vrolife, 2023-11-30:
No

vitalif, 2024-05-25:
@vrolife I've also got Redmibook Pro 15 2023 with Ryzen 7840HS
I wanted to ask how do you create these fixes? )
Redmibook Pro 15 2023 has a problem with the default S0 suspend/resume - after suspending and resuming it doesn't power off correctly i.e. it hangs during poweroff/reboot and you can only turn it off forcibly by holding the power button.
I tried to enable S3 by applying a part of your patch with _S3 (without additional changes) - it didn't help, S3 appeared in dmesg, I ran `echo deep > /sys/power/mem_sleep` and tried to suspend - the laptop suspended but didn't wake up after suspend at all.
I would like to help making a fix for this laptop - could you give me a hint what I should start with?