vrolife/modern_laptop #95: I broke the kernel in an attempt to install a fingerprint on my Redmibook. How can this be fixed?

rybalkoyaa, 2024-03-23:
I broke the kernel in an attempt to install a fingerprint on my Redmibook. How can this be fixed?

```
sudo apt autoremove
Чтение списков пакетов… Готово
Построение дерева зависимостей… Готово
Чтение информации о состоянии… Готово         
Обновлено 0 пакетов, установлено 0 новых пакетов, для удаления отмечено 0 пакетов, и 0 пакетов не обновлено.
Установлено или удалено не до конца 1 пакетов.
После данной операции объём занятого дискового пространства возрастёт на 0 B.
Настраивается пакет linux-image-6.5.0-26-generic (6.5.0-26.26) …
Обрабатываются триггеры для linux-image-6.5.0-26-generic (6.5.0-26.26) …
/etc/kernel/postinst.d/dkms:
 \* dkms: running auto installation service for kernel 6.5.0-26-generic
Sign command: /usr/bin/kmodsign
Signing key: /var/lib/shim-signed/mok/MOK.priv
Public certificate (MOK): /var/lib/shim-signed/mok/MOK.der

Building module:
Cleaning build area...(bad exit status: 2)
make -j16 KERNELRELEASE=6.5.0-26-generic ENTRY=yes all....(bad exit status: 2)
ERROR (dkms apport): binary package for 8852be: 1.0.5 not found
Error! Bad return status for module build on kernel: 6.5.0-26-generic (x86_64)
Consult /var/lib/dkms/8852be/1.0.5/build/make.log for more information.
dkms autoinstall on 6.5.0-26-generic/x86_64 succeeded for redmibook_kbd_backligh
t redmibook_wmi rtw89
dkms autoinstall on 6.5.0-26-generic/x86_64 failed for 8852be(10)
Error! One or more modules failed to install during autoinstall.
Refer to previous errors for more information.
 \* dkms: autoinstall for kernel 6.5.0-26-generic
   ...fail!
run-parts: /etc/kernel/postinst.d/dkms exited with return code 11
dpkg: ошибка при обработке пакета linux-image-6.5.0-26-generic (--configure):
 подпроцесс из пакета linux-image-6.5.0-26-generic установлен сценарий post-inst
allation возвратил код ошибки 1
При обработке следующих пакетов произошли ошибки:
 linux-image-6.5.0-26-generic
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

At the same time, fingerprint did not work, although it was installed. But I would like to, if possible

**Desktop:**
Ubuntu Ubuntu 23.10, IMI Redmi Book Pro 14 2022, 12th Gen Intel® Core™ i7-12650H × 16

_[the translator was used]_

vrolife, 2024-03-24:
You should remove 8852be driver from dkms, it is deprecated. findgerprint driver support USB 10a5:9201 only, check the device installed in your system.

rybalkoyaa, 2024-03-24:
I tried, but I do not know how to delete 8852be correctly...

```
ybalkoyaa@yarbook:~$ cd modern_laptop

rybalkoyaa@yarbook:~/modern_laptop$ sudo /bin/sh uninstall.sh rtl8852be
Сущ:1 http://ru.archive.ubuntu.com/ubuntu mantic InRelease
Сущ:2 http://repo.yandex.ru/yandex-disk/deb stable InRelease                   
Сущ:3 http://ru.archive.ubuntu.com/ubuntu mantic-updates InRelease             
Сущ:4 http://security.ubuntu.com/ubuntu mantic-security InRelease              
Сущ:5 https://repo.yandex.ru/yandex-browser/deb stable InRelease               
Сущ:6 http://ru.archive.ubuntu.com/ubuntu mantic-backports InRelease           
Сущ:7 http://ru.archive.ubuntu.com/ubuntu mantic-proposed InRelease            
Сущ:8 https://ppa.launchpadcontent.net/danielrichter2007/grub-customizer/ubuntu mantic InRelease
Чтение списков пакетов… Готово            
Построение дерева зависимостей… Готово
Чтение информации о состоянии… Готово         
Все пакеты имеют последние версии.
W: http://repo.yandex.ru/yandex-disk/deb/dists/stable/InRelease: Key is stored in legacy trusted.gpg keyring (/etc/apt/trusted.gpg), see the DEPRECATION section in apt-key(8) for details.
Чтение списков пакетов… Готово
Построение дерева зависимостей… Готово
Чтение информации о состоянии… Готово         
Уже установлен пакет dkms самой новой версии (3.0.11-1ubuntu10).
Уже установлен пакет bash самой новой версии (5.2.15-2ubuntu1).
Уже установлен пакет make самой новой версии (4.3-4.1build1).
Уже установлен пакет acpica-tools самой новой версии (20200925-8).
Уже установлен пакет dmidecode самой новой версии (3.5-2).
Уже установлен пакет mokutil самой новой версии (0.6.0-2).
Уже установлен пакет patch самой новой версии (2.7.6-7build2).
Уже установлен пакет diffutils самой новой версии (1:3.8-4).
Обновлено 0 пакетов, установлено 0 новых пакетов, для удаления отмечено 0 пакетов, и 0 пакетов не обновлено.
Установлено или удалено не до конца 1 пакетов.
После данной операции объём занятого дискового пространства возрастёт на 0 B.
Настраивается пакет linux-image-6.5.0-26-generic (6.5.0-26.26) …
Обрабатываются триггеры для linux-image-6.5.0-26-generic (6.5.0-26.26) …
/etc/kernel/postinst.d/dkms:
 \* dkms: running auto installation service for kernel 6.5.0-26-generic
Sign command: /usr/bin/kmodsign
Signing key: /var/lib/shim-signed/mok/MOK.priv
Public certificate (MOK): /var/lib/shim-signed/mok/MOK.der

Building module:
Cleaning build area...(bad exit status: 2)
make -j16 KERNELRELEASE=6.5.0-26-generic ENTRY=yes all...(bad exit status: 2)
ERROR (dkms apport): binary package for 8852be: 1.0.5 not found
Error! Bad return status for module build on kernel: 6.5.0-26-generic (x86_64)
Consult /var/lib/dkms/8852be/1.0.5/build/make.log for more information.
dkms autoinstall on 6.5.0-26-generic/x86_64 succeeded for redmibook_kbd_backligh
t redmibook_wmi rtw89
dkms autoinstall on 6.5.0-26-generic/x86_64 failed for 8852be(10)
Error! One or more modules failed to install during autoinstall.
Refer to previous errors for more information.
 \* dkms: autoinstall for kernel 6.5.0-26-generic
   ...fail!
run-parts: /etc/kernel/postinst.d/dkms exited with return code 11
dpkg: ошибка при обработке пакета linux-image-6.5.0-26-generic (--configure):
 подпроцесс из пакета linux-image-6.5.0-26-generic установлен сценарий post-inst
allation возвратил код ошибки 1
При обработке следующих пакетов произошли ошибки:
 linux-image-6.5.0-26-generic
E: Sub-process /usr/bin/dpkg returned an error code (1)

rybalkoyaa@yarbook:~/modern_laptop$ sudo /bin/sh uninstall.sh blue8852be
Сущ:1 http://ru.archive.ubuntu.com/ubuntu mantic InRelease
Сущ:2 http://repo.yandex.ru/yandex-disk/deb stable InRelease                   
Сущ:3 http://ru.archive.ubuntu.com/ubuntu mantic-updates InRelease             
Сущ:4 http://security.ubuntu.com/ubuntu mantic-security InRelease              
Сущ:5 https://repo.yandex.ru/yandex-browser/deb stable InRelease               
Сущ:6 http://ru.archive.ubuntu.com/ubuntu mantic-backports InRelease           
Сущ:7 http://ru.archive.ubuntu.com/ubuntu mantic-proposed InRelease            
[... 83 lines cut ...]
dkms autoinstall on 6.5.0-26-generic/x86_64 failed for 8852be(10)
Error! One or more modules failed to install during autoinstall.
Refer to previous errors for more information.
 * dkms: autoinstall for kernel 6.5.0-26-generic
   ...fail!
run-parts: /etc/kernel/postinst.d/dkms exited with return code 11
dpkg: ошибка при обработке пакета linux-image-6.5.0-26-generic (--configure):
 подпроцесс из пакета linux-image-6.5.0-26-generic установлен сценарий post-inst
allation возвратил код ошибки 1
При обработке следующих пакетов произошли ошибки:
 linux-image-6.5.0-26-generic
E: Sub-process /usr/bin/dpkg returned an error code (1)
```
Yes, I have the same device

```
balkoyaa@yarbook:~/modern_laptop$ lsusb
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 003: ID 10a5:9201 FPC FPC Sensor Controller L:0001 FW:021.26.2.031
Bus 003 Device 002: ID 1bcf:2d04 Sunplus Innovation Technology Inc. XiaoMi USB 2.0 Webcam
Bus 003 Device 004: ID 0cb8:c559 Opticis Co., Ltd Bluetooth Radio
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

vrolife, 2024-03-25:
To remove 8852be
`
sudo dkms remove 8852be/1.0.5 --all
`

rybalkoyaa, 2024-03-26:
All right, thank you, that was very helpful.

And if there is an opportunity, then I would still like fingerprint to turn on.

vrolife, 2024-03-27:
Try to insatll fingerprint driver from this branch https://github.com/vrolife/modern_laptop/tree/static_ocv_fingerprint_driver