vrolife/modern_laptop #96: Help with TM2309-Redmi_Book_Pro_16_2024

uselessfire, 2024-06-27:
Hello. Can you please help with TM2309-Redmi_Book_Pro_16_2024?
BIOS version RMAMT6B0P0808
It has the same exception with dmic (sound works only with dmic_detect=0, but not microphone).

I tried to patch my ACPI table like your script do (part with AcpDmicConnected), but faced into problems.

I dumped my ACPI tables with `acpidump -b` ([acpi_dump.zip](https://github.com/user-attachments/files/16009532/acpi_dump.zip))

Then tried to disassemble sorted `ssdt*.dat` files like your script do, but it fails with error:

`Firmware Error (ACPI): Failure creating named object [\_SB.PC00.XHCI.RHUB.HS03._UPC], AE_ALREADY_EXISTS (20230628/dswload-495)`
`ACPI Error: AE_ALREADY_EXISTS, During name lookup/catalog (20230628/psobject-372)`
`Could not parse ACPI tables, AE_ALREADY_EXISTS`

After disassembling each file apart, i found that `_SB_.PC00.XHCI.RHUB.HS03` device can be found in `ssdt24.dat`, and tried to disassemble all `ssdt*.dat` files without `ssdt24.dat`.
It works, i received [dsdt.dsl.zip](https://github.com/user-attachments/files/16009882/dsdt.dsl.zip) file which not contains `ssdt24.dat`.

Then i patched it by myself with various patches from your repository
More specifically i done:
`-DefinitionBlock ("", "DSDT", 1, "XMCC  ", "XMCC2019", 0x00000002)`
`+DefinitionBlock ("", "DSDT", 1, "XMCC  ", "XMCC2019", 0x00000003)`
And added
`+                    Name (_DSD, Package (0x02)  // _DSD: Device-Specific Data`
`+                    {`
`+                        ToUUID ("5025030f-842f-4ab4-a561-99a5189762d0") /* Unknown UUID */,`
`+                        Package (0x01)`
`+                        {`
`+                            Package (0x02)`
`+                            {`
`+                                "AcpDmicConnected",`
`+                                One`
`+                            }`
`+                        }`
`+                    })`
to two `HS05` device blocks (i can't found `ACP` device, but `HS05` device has the same address by `Name (_ADR, 0x05)  // _ADR: Address`, like in your patches)

Then i tried to compile resulted `dsdt.dsl` with `iasl -ve dsdt.dsl`, but it fails with error:
`dsdt.dsl    743:     External (_SB_.PC00.XHCI._PS0.PS0X, MethodObj)    // 0 Arguments`
`Error    6163 -                                      ^ Object is created temporarily in another method and cannot be accessed (_SB_.PC00.XHCI._PS0.PS0X)`

`dsdt.dsl    744:     External (_SB_.PC00.XHCI._PS3.PS3X, MethodObj)    // 0 Arguments`
`Error    6163 -                                      ^ Object is created temporarily in another method and cannot be accessed (_SB_.PC00.XHCI._PS3.PS3X)`

`dsdt.dsl   1072:     External (_SB_.PC02.XHCI._PS0.PS0X, MethodObj)    // 0 Arguments`
`Error    6163 -                                      ^ Object is created temporarily in another method and cannot be accessed (_SB_.PC02.XHCI._PS0.PS0X)`

`dsdt.dsl   1073:     External (_SB_.PC02.XHCI._PS3.PS3X, MethodObj)    // 0 Arguments`
`Error    6163 -                                      ^ Object is created temporarily in another method and cannot be accessed (_SB_.PC02.XHCI._PS3.PS3X)`

`ASL Input:     dsdt.dsl - 3274817 bytes  56426 keywords      0 source lines`

`Compilation failed. 4 Errors, 88 Warnings, 1135 Remarks`
`No AML files were generated due to compiler error(s)`

Also tried `iasl -va dsdt.dsl`, the same error.

**Can you please help with this laptop? I ready to provide all information that you need. Also can pay for it if needed (email me by uselessfire@gmail.com).**

Laptop also has issue with multimedia keys on F7, F8, F9, and "settings" button on the top right edge of the keyboard, it all provides the same ACPI event `wmi PNP0C14:00 00000020 00000000`.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
Arch Linux
 - Kernel version: `uname -a`
Linux wildarch 6.9.6-arch1-1 SMP PREEMPT_DYNAMIC Fri, 21 Jun 2024 19:49:19 +0000 x86_64 GNU/Linux
 - Environment: `cat /etc/os-release`
NAME="Arch Linux"
PRETTY_NAME="Arch Linux"
ID=arch
BUILD_ID=rolling
ANSI_COLOR="38;2;23;147;209"
HOME_URL="https://archlinux.org/"
DOCUMENTATION_URL="https://wiki.archlinux.org/"
SUPPORT_URL="https://bbs.archlinux.org/"
BUG_REPORT_URL="https://gitlab.archlinux.org/groups/archlinux/-/issues"
PRIVACY_POLICY_URL="https://terms.archlinux.org/docs/privacy-policy/"
LOGO=archlinux-logo

** Log **
[dmesg.log](https://github.com/user-attachments/files/15982958/dmesg.log)

vrolife, 2024-06-27:
AcpDmicConnected should only be effective on AMD's acp6x platforms, whereas your CPU is from Intel.

uselessfire, 2024-06-27:
Okay, so can you help me? Issue seems to be similar. I tried many `snd-hda-intel model=` and other module parameters, various kernels and versions, various distributives, none of them worked. It be very helpful for community of Redmibook Pro users on Linux.

vrolife, 2024-06-27:
I might not be able to assist much without the hardware to reproduce this issue.

uselessfire, 2024-06-27:
Just booted Windows and installed all updated, including BIOS firmware. Now it's version is RMAMT6B0P0909, but the issue is still here

vrolife, 2024-06-28:
Find some useful information
https://tech-docs.system76.com/models/lemp13/README.html#lemur-pro-lemp13
https://github.com/system76/firmware-open/blob/master/models/lemp13/hda_verb.c

uselessfire, 2024-06-28:
It's for ALC245, but i have 8086:7e28 ALC256.

uselessfire, 2024-06-28:
Also i tried Pop_OS! by System76, microphone doesn't work with or without `snd-hda-intel dmic_detect=0` and `snd-intel-dspcfg dsp_driver=1`

metajiji, 2024-11-24:
> Just booted Windows and installed all updated, including BIOS firmware. Now it's version is RMAMT6B0P0909, but the issue is still here

[uselessfire](https://github.com/uselessfire), hello, have you `RMAMT6B0P0909.exe` file?

I upgrade BIOS to `RMAMT6B0P0A0A.exe` and linux kernel from grub wont start anymore, system will hang permanently and i see black screen with only `_` symbol.

othercat, 2024-12-05:
> [uselessfire](https://github.com/uselessfire), hello, have you `RMAMT6B0P0909.exe` file?
>
> I upgrade BIOS to `RMAMT6B0P0A0A.exe` and linux kernel from grub wont start anymore, system will hang [...]

I've downloaded the RMAMT6B0P0909 from the link: https://4pda.to/forum/index.php?showtopic=1085675&st=2180
And I just unzip the 7z file and re-compress to zip file which the format is allowd by Github.
Here you go~
[RMAMT6B0P0909.zip](https://github.com/user-attachments/files/18016847/RMAMT6B0P0909.zip)
sha256checksum=d802aeea4b2c8251d9b0ba8187461feba47efb37885ddad800a1eed9c310a725

othercat, 2024-12-05:
I've just downgraded the BIOS, and everything has returned to normal~
[image]

uselessfire, 2025-01-12:
Microphone works from version 2024.06 of sof-firmware, see https://github.com/uselessfire/RedmiBook-Pro-14-2024/#installing-linux

closed 2025-01-12 by uselessfire

NMNMCC, 2025-04-09:
@uselessfire Hello, I am experiencing the same problem as you, but I cannot find the old BIOS for downgrading. Could you please upload it? If you can, I will be very grateful!

othercat, 2025-04-09:
I can't find the original post from your request. Do u still need the
downgrading firmware, and what's the model of your laptop

metajiji, 2025-04-09:
> I can't find the original post from your request. Do u still need the
> downgrading firmware, and what's the model of your laptop
> […](#)

https://www.catalog.update.microsoft.com/Search.aspx?q=xiaomi%20Firmware

All firmwares here

You can download oficial xiaomi tool, extract it (sfx 7z), and replace old firmware from MS site, in my case `TM2309.fd`.

And change `platform.ini` for disable version check and allow downgrade

```ini
[UpdateOEMME]
CheckVersion=0
```

NMNMCC, 2025-04-09:
@othercat Hello, I'm looking for the BIOS of RedmiBook Pro 14 2024 (I was mistakenly thinking that the same firmware as the 16-inch model, so I deleted the post)

NMNMCC, 2025-04-09:
@metajiji Thank you so much for the information, I am trying what you said!

NMNMCC, 2025-04-09:
@othercat @metajiji 

[image]
[image]

NMNMCC, 2025-04-12:
[image]